stm32-rs/stm32g4xx-hal #1: What would be reqired to get this to a working state?

bentwire, 2019-08-04:
I'm interested in getting USB up and running on these micros, so I got myself a STM32G474 Nucleo and a STM32G431 Nucleo board.  Was trying to get the blinky example to work but it won't compile.  

I was wondering how much work was required to get this to a working state and if someone could possibly point me in the right direction.

dotcypress, 2019-11-25:
@bentwire sorry again about PR. 
I'm working mostly on G0 HAL, and sometimes on this one.
Now you can compile examples from `master` branch.

Feel free to send any PR needed, but make them small and grained please.

bentwire, 2019-11-26:
Ok, I can split up what I have in to functional unit chunks and resubmit as multiple prs if you would like.

yehielliv, 2020-06-29:
I try to implement by using
cargo generate --git https://github.com/stm32-rs/stm32g4xx-hal

Then when I did cargo build I got this error
Compiling typenum v1.11.2
   Compiling proc-macro2 v1.0.6
   Compiling semver-parser v0.7.0
   Compiling unicode-xid v0.2.0
   Compiling rand_core v0.4.2
   Compiling syn v1.0.8
   Compiling cortex-m v0.6.2
   Compiling cortex-m-rt v0.6.10
   Compiling stm32g4 v0.9.0
   Compiling semver v0.9.0
   Compiling rand_core v0.3.1
   Compiling rand v0.5.6
error: failed to run custom build command for `stm32g4 v0.9.0`

Caused by:
  process didn't exit successfully: `C:\Users\KNC\workspace_rust\stm32-g4\target\debug\build\stm32g4-e59bfc4da205d66f\build-script-build` (exit code: 101)
--- stdout
cargo:rustc-link-search=C:\Users\KNC\workspace_rust\stm32-g4\target\thumbv7em-none-eabihf\debug\build\stm32g4-d469d43ef2d76064\out

--- stderr
thread 'main' panicked at 'No device features selected', C:\Users\KNC\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32g4-0.9.0\build.rs:22:18
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

warning: build failed, waiting for other jobs to finish...
error: build failed

Please let me know if there is a way to fix it.

Thank you.

barafael, 2020-07-31:
For that specific error, see issue #3

luctius, 2021-07-29:
@bentwire Is this issue still relevant?

bentwire, 2021-07-29:
I've not had a chance to revisit this in quite awhile.  Close it if you want, I assume the state of the crate has changed quite a bit since I opened this.

closed 2021-07-29 by luctius