stm32-rs/stm32g4xx-hal #3: Follow instructions did not work

yehielliv, 2020-06-30:
I try to implement by using
cargo generate --git https://github.com/stm32-rs/stm32g4xx-hal

Then when I did cargo build I got this error
Compiling typenum v1.11.2
   Compiling proc-macro2 v1.0.6
   Compiling semver-parser v0.7.0
   Compiling unicode-xid v0.2.0
   Compiling rand_core v0.4.2
   Compiling syn v1.0.8
   Compiling cortex-m v0.6.2
   Compiling cortex-m-rt v0.6.10
   Compiling stm32g4 v0.9.0
   Compiling semver v0.9.0
   Compiling rand_core v0.3.1
   Compiling rand v0.5.6
error: failed to run custom build command for `stm32g4 v0.9.0`

Caused by:
  process didn't exit successfully: `C:\Users\KNC\workspace_rust\stm32-g4\target\debug\build\stm32g4-e59bfc4da205d66f\build-script-build` (exit code: 101)
--- stdout
cargo:rustc-link-search=C:\Users\KNC\workspace_rust\stm32-g4\target\thumbv7em-none-eabihf\debug\build\stm32g4-d469d43ef2d76064\out

--- stderr
thread 'main' panicked at 'No device features selected', C:\Users\KNC\.cargo\registry\src\github.com-1ecc6299db9ec823\stm32g4-0.9.0\build.rs:22:18
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

warning: build failed, waiting for other jobs to finish...
error: build failed

Please let me know if there is a way to fix it.

Thank you.

barafael, 2020-07-31:
This crate has a number of devices which are supported. These are exposed as feature flags. You need to state the device you want to target in Cargo.toml.

I did it this way: 

[features]
default = ["rt"]

becomes:

[features]
default = ["rt", "stm32g474"]

I think this is not a good way to do it (but ok for test). Ideally one would add this crate as a dependency and then state the feature flags in the project Cargo.toml.

luctius, 2021-07-29:
@yehielliv Is this issue stil relevant?

closed 2021-08-13 by luctius