stm32-rs/stm32g4xx-hal #10: GPIO alternate function not configurable

Geens, 2021-03-21:
It's  not possible to configure the alternate function registers of the GPIO pins. This is needed for implementing peripherals like I2C and others.

The STM32F3 series has the same register interface for GPIO. I'll try and port their [implementation](https://github.com/stm32-rs/stm32f3xx-hal/blob/master/src/gpio.rs) and [code generator ](https://github.com/stm32-rs/stm32f3xx-hal/tree/master/codegen). I'm making this issue for tracking and discussion.

Geens, 2021-04-05:
It seems like this is already possible but I just missed it in the docs because of the pub(crate) access of the set_alt_mode function. The #[allow(dead_code)] on the AltFunction enum made me think it was just a stub and I didn't look any further.

I have already done some work on implementing this using typestates, like the stm32f3xx_hal crate does. I think this is a cleaner approach and will make porting other peripherals already implemented in stm32f3xx_hal easier.

My main goal is to get I2C to work. It seems like the stm32g4 and stm32f3 series are very similar in everything that's needed to do so. Would it be useful to share more of the code between both platforms? Is this the right place to discuss this?

luctius, 2021-07-29:
This should be fixed by #83e71c6.

closed 2021-07-29 by luctius