stm32-rs/stm32g4xx-hal #98: Timer driven Events

no111u3, 2023-12-02:
There aren't any way to setup run ADC from any timer or other event.
Also there aren't any functions to create chain of timers.

usbalbin, 2023-12-02:
[Here](https://github.com/stm32-rs/stm32g4xx-hal/pull/96/files#diff-eac3481d9342de2b4e8bdf87136844f041041815b5065a5d712c743e139dd2af) is one way in my very much work in progress hrtim PR. :)

usbalbin, 2024-01-01:
Is that sort of API something we want to use for most event based things?

## Just to summarize:

Avoid enums for representing event sources in the user facing api by instead letting them use the relevant object itself directly (or a reference to it). This also has the side effects that the event source needs to be setup before the consumer starts listening.

### For example:

```rust
let (timer, (cr1, ..), (mut out1, mut out2)) = dp
        .HRTIM_TIMA
        .pwm_advanced((pin_a, pin_b), &mut rcc)
        [...]
        .finalize(&mut hr_control);

    //Good
    out1.enable_rst_event(&cr1); // Set low on compare match with cr1
    out1.enable_set_event(&timer); // Set high at new period
   
    // Bad
    out1.enable_rst_event(HrTimerEvent::Cr1); // Set low on compare match with cr1
    out1.enable_set_event(HrTimerEvent::Period); // Set high at new period
```

---

Incomplete list of similar things:

* Comp inverting input: pin, dac or vref division (not events but still a bit similar)
* Opamp: I believe same as comp
* DAC trigger: timers and hrtim I believe
* Adc trigger source: lots of timers, hrtim "trigger sources", etc.
* Hrtim output SET/RST: compare match, period, same things from other timers, hrtim "external events" etc.
* hrtim period: about the same as above I believe
* hrtim fault: comp, digital pin etc.
* hrtim "external event": similar as above

usbalbin, 2024-01-05:
So implementation wise that would mean that enums are avoided and instead traits are used with member const that are used to configure the listener.

Does this make sense? If so, would you like me to update #96 to follow that pattern? Or does anyone have other thoughts? :)