stm32-rs/stm32g4xx-hal #184: stm32g491 not compiling

JDRobotter, 2025-02-17:
It seems that this crate does not compile using stm32g491 feature, a lot of unresolved import, redefinitions, etc.
I'm ready to do what's necessary to patch this myself but I need some pointers. Can anybody point me to relevant issues or procedure to fix this crate for stm32g491 ?

I'm including the first lines of compile log here:
```
error[E0432]: unresolved import `crate::stm32::FDCAN2`
   --> src/can.rs:128:9
    |
128 |     use crate::stm32::FDCAN2;
    |         ^^^^^^^^^^^^^^------
    |         |             |
    |         |             help: a similar name exists in the module: `FDCAN`
    |         no `FDCAN2` in `stm32g491`

error[E0432]: unresolved import `crate::opamp::OpampEx`
  --> src/prelude.rs:36:9
   |
36 | pub use crate::opamp::OpampEx as _;
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^ no `OpampEx` in `opamp`

error[E0432]: unresolved import `crate::stm32::TIM20`
   --> src/pwm.rs:185:5
    |
185 | use crate::stm32::TIM20;
    |     ^^^^^^^^^^^^^^-----
    |     |             |
    |     |             help: a similar name exists in the module: `TIM2`
    |     no `TIM20` in `stm32g491`

error[E0412]: cannot find type `Opamp1` in module `opamp`
    --> src/adc.rs:2821:12
     |
2821 |     opamp::Opamp1 => (ADC1, 13),
     |            ^^^^^^ not found in `opamp`
...
```

usbalbin, 2025-02-17:
Is this on latest master?

usbalbin, 2025-02-17:
If so, would you like to check if this works better on the [staged-pac](https://github.com/stm32-rs/stm32g4xx-hal/tree/staged-pac) branch. I believe there might be some fixes in the new pac that are only present in that branch

JDRobotter, 2025-02-18:
Yes it was on master !
I've successfully compiled and tested some code on target for my stm32g491 using the stage-pac branch.

closed 2025-04-12 by JDRobotter

usbalbin, 2025-04-12:
Unless someone objects, my plan is to merge that branch into master once all PRs targeting that are closed.