stm32-rs/stm32g4xx-hal #208: Default pin mode

burrbull, 2025-05-28:
@usbalbin According to RM default pin mode (state on reset) for G4 is `Analog`.

https://github.com/stm32-rs/stm32g4xx-hal/blob/64aa1ff0eddcd4a0532886d415a1ecdfc933e337/src/gpio.rs#L11

usbalbin, 2025-05-28:
Yes
> (7/7) running `dac`...
ERROR panicked at tests/nucleo-g474.rs:358:9:
assertion failed: !is_pax_low(gpioa, 4)

usbalbin, 2025-05-28:
Moving the dac test to `tests/nucleo-g474_w_jumpers.rs` and instead checking `pa1` (which should be connected to pa4 for those tests) and it works fine again

AdinAck, 2025-05-29:
> Reset value: 0xABFF FFFF (for port A)

Yeah those pins on port A are for the SWD pins, by default they are in the respective alternate function mode (for good reason, of course).