stm32-rs/stm32g4xx-hal #211: unconstrained opaque types  impl SDAPin<I2C1>, impl SCLPin<I2C1> with updated Rust

pdgilbert, 2025-06-22:
I have several examples using a setup function in my crate's  `src/setup_all_stm32g4xx.rs` file. The setup functions returns a type defined by
```
pub type I2c1Type = I2c<I2C1, impl SDAPin<I2C1>, impl SCLPin<I2C1>>;
```

With `rustc 1.85.0-nightly` from 2025-01-01  the file `src/setup_all_stm32g4xx.rs` and most of the examples compile (75 of 84). With the recent`rustc 1.89.0-nightly`  I get

```
error: unconstrained opaque type
  --> src/setup_all_stm32g4xx.rs:50:31
   |
50 | pub type I2c1Type = I2c<I2C1, impl SDAPin<I2C1>, impl SCLPin<I2C1>>;
   |                               ^^^^^^^^^^^^^^^^^
   |
   = note: `I2c1Type` must be used in combination with a concrete type within the same crate

error: unconstrained opaque type
  --> src/setup_all_stm32g4xx.rs:50:50
   |
50 | pub type I2c1Type = I2c<I2C1, impl SDAPin<I2C1>, impl SCLPin<I2C1>>;
   |                                                  ^^^^^^^^^^^^^^^^^
   |
   = note: `I2c1Type` must be used in combination with a concrete type within the same crate

error: unconstrained opaque type
  --> src/setup_all_stm32g4xx.rs:51:31
   |
51 | pub type I2c2Type = I2c<I2C2, impl SDAPin<I2C2>, impl SCLPin<I2C2>>;
   |                               ^^^^^^^^^^^^^^^^^
   |
   = note: `I2c2Type` must be used in combination with a concrete type within the same crate

error: unconstrained opaque type
  --> src/setup_all_stm32g4xx.rs:51:50
   |
51 | pub type I2c2Type = I2c<I2C2, impl SDAPin<I2C2>, impl SCLPin<I2C2>>;
   |                                                  ^^^^^^^^^^^^^^^^^
   |
   = note: `I2c2Type` must be used in combination with a concrete type within the same crate

error[E0308]: mismatched types
   --> src/setup_all_stm32g4xx.rs:176:10
    |
50  | ...type I2c1Type = I2c<I2C1, impl SDAPin<I2C1>, impl SCLPin<I2C1>>;
    |                              -----------------  ----------------- one of the expected opaque types
    |                              |
    |                              one of the expected opaque types
...
176 | ...pin, i2c1, i2c2, led, tx1, rx1,  tx2, rx2, spi1, spiext,  delay, clo...
    |         ^^^^ expected opaque type, found `Pin<'B', 9, Alternate<4, OpenDrain>>`
    |
    = note: expected struct `stm32g4xx_hal::i2c::I2c<_, setup_all_stm32g4xx::I2c1Type::{opaque#0}, setup_all_stm32g4xx::I2c1Type::{opaque#1}>`
               found struct `I2c<Periph<..., 1073763328>, ..., ...>`
note: this item must have a `#[define_opaque(setup_all_stm32g4xx::I2c1Type)]` attribute to be able to define hidden types
...
```

I have made some attempts with  `#[define_opaque(...)]` but no luck.

When the `src/` file does not compile none of the examples compile. I can resolve this by going back to a concrete type definition, but I think I should not need to do that. I can also resolve it by using the old compiler, but I assume the new compiler is complaining about something that really is a problem.

I would appreciate if anyone can explain what the newer Rust is complaining about.

[Hals `stm32f4xx_hal` and `stm32f1xx_hal` avoid the problem by not requiring pins in `I2c<I2C2>`.]

More complete details are available at https://github.com/pdgilbert/rust-integration-testing/actions/runs/15769713028/job/44452339985  (Any of the  `weact-stm32g431` jobs. They all fail for the same reason.)

usbalbin, 2025-06-23:
Have not used that feature but here seems to be some info maybe?https://www.reddit.com/r/rust/s/cr2M0hcx6W

pdgilbert, 2025-06-24:
Thanks @usbalbin , that solves it.

closed 2025-06-24 by pdgilbert