stm32-rs/stm32g4xx-hal #222: spi.rs needs work

phansel, 2025-07-25:
https://github.com/stm32-rs/stm32g4xx-hal/blob/main/src/spi.rs

- embedded_hal_old has not been fully removed / replaced
- hal_02 is still used in [the spi example](https://github.com/stm32-rs/stm32g4xx-hal/blob/main/examples/spi-example.rs#L58)
- little to no explanatory comments (compare: [stm32h7xx-hal](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/spi.rs))
  - unclear what functions like [set_bidi()](https://github.com/stm32-rs/stm32g4xx-hal/blob/ee7c0103f0eb3c8617478646826e785435692c83/src/spi.rs#L204) actually do; do they convert it to full-duplex, or to half-duplex on a 3-wire interface?
- same issue: [many instances](https://github.com/stm32-rs/stm32g4xx-hal/blob/ee7c0103f0eb3c8617478646826e785435692c83/src/spi.rs#L317) of names abbreviated beyond being comprehensible

So far, I haven't been able to correctly read+write to an SPI device from stm32g474 with this HAL yet. 
- Responses (on MISO) to writes (on MOSI) are either not captured by the device or are captured incorrectly. 
- Reads, when they don't crash, make little temporal sense
  - 0x22 on the bus -> reports as [u8; 3] as [0x0c, 0x0c, 0x0c]. 
- With certain orders of operation, the microcontroller spontaneously panics.
- transfer() does not work at all. SCK sits at idle the entire time.

The intuitive API (spi.read(), spi.write()) type system is also broken in a way I have never seen before. rustc gets confused by spi.read() and isn't sure if it's a trait of the ADC, I2C, or different types of SPI.
`spi.read(&mut out_buf)` has to be fully specified as `stm32g4xx_hal::prelude::SpiBus::read` to even compile, and even then, the SPI read behavior is unreliable.
```
error[E0034]: multiple applicable items in scope
   --> src/main.rs:147:9
    |
147 |     spi.read(&mut read_buf);
    |         ^^^^ multiple `read` found
    |
note: candidate #1 is defined in the trait `_embedded_hal_adc_OneShot`
   --> /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-0.2.7/src/adc.rs:97:5
    |
97  |     fn read(&mut self, pin: &mut Pin) -> nb::Result<Word, Self::Error>;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: candidate #2 is defined in the trait `stm32g4xx_hal::prelude::I2c`
   --> /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-1.0.0/src/i2c.rs:333:5
    |
333 |     fn read(&mut self, address: A, read: &mut [u8]) -> Result<(), Self::Error> {
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
note: candidate #3 is defined in the trait `stm32g4xx_hal::prelude::SpiBus`
   --> /home/user/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embedded-hal-1.0.0/src/spi.rs:446:5
    |
446 |     fn read(&mut self, words: &mut [Word]) -> Result<(), Self::Error>;
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    = note: candidate #4 is defined in an impl of the trait `_embedded_hal_spi_FullDuplex` for the type `Spi<SPI, PINS>`
help: disambiguate the method for candidate #1
    |
147 -     spi.read(&mut read_buf);
147 +     _embedded_hal_adc_OneShot::read(&mut spi, &mut read_buf);
    |
help: disambiguate the method for candidate #2
    |
147 -     spi.read(&mut read_buf);
147 +     stm32g4xx_hal::prelude::I2c::read(&mut spi, &mut read_buf);
    |
help: disambiguate the method for candidate #3
    |
147 -     spi.read(&mut read_buf);
147 +     stm32g4xx_hal::prelude::SpiBus::read(&mut spi, &mut read_buf);
    |
help: disambiguate the method for candidate #4
    |
147 -     spi.read(&mut read_buf);
147 +     _embedded_hal_spi_FullDuplex::read(&mut spi, &mut read_buf);
```

usbalbin, 2025-07-25:
cc @techmccat

phansel, 2025-07-25:
I'll take a look at patching a few of these in time. Mega perplexed by the last one

techmccat, 2025-07-25:
Regarding the overrun behavior, I've confirmed that it happens every time I run the test, and that it was working at least up until commit 1b7a8b4, then I guess I messed it up while updating my fork

phansel, 2025-07-25:
Your fork as of f5a2865b5149294afb2d3b5f1cfd98526a58ad89 seems to perform more reliably -- it reads more than one byte mostly correctly (though my BMI323 apparently has a CHIP_ID of 0x44 instead of 0x43). 

The clock timing on reads is still odd, though. At 1 MHz, there's a 20us gap between the first two read bytes, then a 75us gap between the second and third. Bytes from the third onward are followed by a uniform 40us gap, up to the maximum of 20 bytes read. Any extra bytes after index 19 in the read buffer are zeroed out.

The last issue I mentioned (conflicting methods) is still present. For reference, these are the imports I'm using:

```
use cortex_m_rt::entry;
use stm32g4xx_hal as hal;
use defmt::{Format, debug, error, info, trace, warn};
use defmt_rtt as _;
use hal::{
    delay::DelayFromCountDownTimer,
    gpio::{AF5, PA5, PA6, PA7},
    prelude::*,
    pwr::PwrExt,
    rcc::Config,
    spi,
    stm32::Peripherals,
    time::{ExtU32, RateExtU32},
    timer::Timer,
};
```

techmccat, 2025-07-26:
I'm not too surprised by the timing since it's affected by the FIFO prefill behavior (prefill write fifo, then start read/write loop), though it's a bit odd that it performs that bad at 1MHz, in my benchmarks it could run at full speed 4MHz with the core clocked at 16MHz from the HSI. Have you build your code in debug mode?

The 20 byte limit is concerning but i'm unable to reproduce it when running the spi-hal-one example on a Nucleo G474RE with MOSI and MISO shorted together, doesn't help that I can't get the rp2040 I'm using as a logic analyzer to sample this spi traffic correctly
I've done further tests with my ST25R39 driver still seems to reliably detect and activate NFC-A cards too.

Regarding the conflicting methods issue, could you upload a full program with that issue? I've tried making a separate crate and running the example there as bin but the read/write/transfer methods are found just fine by the compiler

phansel, 2025-07-26:
It was built in debug mode to comply with defmt -- I could see that affecting the timings. Will re-examine in release mode.

For the logic analyzer, it might help to drop the STM32 bus rate. I can see on my end that SCLK only clocks out 20 bytes when read() is given a 40 byte slice to fill.

Here's a minimum repro for rustc 1.90.0-nightly and 1.88.0-stable: [stm32g4xx-hal-spi-demo](https://github.com/phansel/stm32g4xx-hal-spi-demo).

techmccat, 2025-07-26:
I think I've figured out the issue with your imports, you're importing the old FullDuplex trait [on line 63](https://github.com/phansel/stm32g4xx-hal-spi-demo/blob/4779f4cda1bdd1735786c4881aef2770cc56db8a/src/main.rs#L63), which means you now have two traits implementing a read() on spi in scope.

I've removed the import and called hal::hal_02::spi::FullDuplex::send directly and it seems to work fine, I can't see the limit to 20 read bytes behavior, though I think alternating between the old and new trait functions might break something since I didn't think about them being used together when I modified the driver

This is the output I'm getting from your program, after editing the loop to log all the read data on a single line for convenience

```
INFO  Booted up.
INFO  pre-read rbuf hex: [ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff, ff]:
INFO  Set CS low
INFO  LSbytes of rbuf: 0x00 0x00 0x00
INFO  Full rbuf: [00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00, 00]
INFO  set cs high
```

I've also tested it with MISO connected to 3v3 and it fills with 0xff as it should

techmccat, 2025-07-28:
If I understand the SpiBus trait correctly, that should be the intended behavior: transfer does reading and writing at the same time, while write + read does it sequentially, so the value read by transfer is offset by 1 because you wrote one byte on the bus and the SPI slave was holding the line low.
As far as I understand it's meant for SPI devices that can do read while write.

phansel, 2025-07-28:
@techmccat good catch! Seems to be right. I might have caught that if I checked with a larger write buffer.